ros-drivers/microstrain_mips #1: Add a parameter to allow operation with -25 devices

jeff-o, 2017-04-06:
Hi, is there a way to use this driver with the 3DM-GX5-25 device? I find that the driver launches just fine, but does not publish the IMU data. 

Thanks!

bsb808, 2017-04-06:
Yes.  I have some non ROS code I can port over to this package. Would you
be willing to test with the 25 hardware? I don't have the device to be able
to verify the functionality.

Brian

jeff-o, 2017-04-07:
Yup, I have one here, ready for any testing you require!

bsb808, 2017-04-08:
> Phone: 519-513-2416 <(519)%20513-2416>

Great. Will take me a couple days to get to it, but I'll let you know when
we have a prototype to test.
Brian

bsb808, 2017-04-11:
Jeff et al,

I reorganized the source and added a -25 specific launch file.  Please use the "gx25" branch to test with the device.  

In the launch file I turned off both the GPS messages and the Filter (KF) messages.  With these turned off the driver does not try to setup the GPS and KF functionality - which the -25 doesn't have.

My understanding is that the -25 uses a complementary filter for the IMU data, so the reset_filter service should/may reset the CF.

If you can test the branch with the hardware it would be much appreciated.

bsb808, 2017-04-11:
PS - You should be able to test with the microstrain_25.launch file.

jeff-o, 2017-04-11:
First test looks pretty good! Pasted below for your reference are the first few lines on startup. I can push my changes for the launch file if that would be helpful, but it's very minor stuff.

Some things I noticed:
- x, y, z acceleration are reported properly. Z orientation seems stable, but takes a few moments to stabilize after the device is swung around at high speed.
- when canceled the node escalates to SIGTERM every time.
- there are references to "gx4" in the launch file that are easily fixed.
- on launch the node still reports "enabling GPS stream" though there is no GPS topic published.
- it seems to be nice and stable at 100Hz, nice!
- resource usage seems high at 100% CPU, but only 0.7% memory. Causes my computer fans to spin up.
- all covariance values are 0.0. I'm not sure we can trust it THAT much ;)
- for the sake of differentiating from other potential onboard IMU devices, make the topic name a parameter and/or change the default to something more descriptive like "gx5-imu"

I can record a rosbag if that would be helpful. What else can I help with?

SUMMARY
========

PARAMETERS
 * /microstrain_3dm_gx4_25_node/auto_init: True
 * /microstrain_3dm_gx4_25_node/baudrate: 115200
 * /microstrain_3dm_gx4_25_node/declination: 0.23
 * /microstrain_3dm_gx4_25_node/declination_source: 2
 * /microstrain_3dm_gx4_25_node/device_setup: True
 * /microstrain_3dm_gx4_25_node/dynamics_mode: 1
 * /microstrain_3dm_gx4_25_node/gps_frame_id: wgs84
 * /microstrain_3dm_gx4_25_node/gps_rate: 4
 * /microstrain_3dm_gx4_25_node/imu_frame_id: base_link
 * /microstrain_3dm_gx4_25_node/imu_rate: 10
 * /microstrain_3dm_gx4_25_node/nav_rate: 10
 * /microstrain_3dm_gx4_25_node/odom_child_frame_id: base_link
 * /microstrain_3dm_gx4_25_node/odom_frame_id: wgs84
 * /microstrain_3dm_gx4_25_node/port: /dev/ttyACM0
 * /microstrain_3dm_gx4_25_node/publish_gps: False
 * /microstrain_3dm_gx4_25_node/publish_imu: True
 * /microstrain_3dm_gx4_25_node/publish_odom: False
 * /microstrain_3dm_gx4_25_node/readback_settings: True
 * /microstrain_3dm_gx4_25_node/save_settings: True
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    microstrain_3dm_gx4_25_node (microstrain_3dm_gx5_45/microstrain_3dm_gx5_45_node)
    rqt_topic (rqt_topic/rqt_topic)

auto-starting new master
process[master]: started with pid [10984]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to af85e04c-1ee1-11e7-8ce2-001c429d378b
process[rosout-1]: started with pid [10997]
started core service [/rosout]
process[microstrain_3dm_gx4_25_node-2]: started with pid [11011]
process[rqt_topic-3]: started with pid [11015]
[ INFO] [1491934032.492565975]: Attempting to open serial port </dev/ttyACM0> at <115200> 

[ INFO] [1491934032.620500252]: Putting device communications into 'standard mode'
[ INFO] [1491934032.627595762]: Verify comm's mode
[ INFO] [1491934032.629381463]: Sleep for a second...
WARNING: Package "ompl" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
WARNING: Package "ompl" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
[ INFO] [1491934033.629567118]: Right mode?
[ INFO] [1491934033.629610365]: Idling Device: Stopping data streams and/or waking from sleep
[ INFO] [1491934034.831352145]: AHRS Base Rate => 1000 Hz
[ INFO] [1491934035.831630423]: Setting the AHRS message format
[ INFO] [1491934037.073590590]: Poll AHRS data to verify
[DEBUG] [1491934037.219430800]: 0 FILTER (0 errors)    1 AHRS (0 errors)    0 GPS (0 errors) Packets
[ INFO] [1491934038.220144214]: Saving AHRS data settings
[ INFO] [1491934039.407932901]: Setting declination source to 0X02
[ INFO] [1491934040.542217973]: Reading back declination source
[ INFO] [1491934040.639355452]: Success: Declination source set to 0X02
[ INFO] [1491934041.639735069]: Saving declination source settings to EEPROM
[ INFO] [1491934042.773617060]: Setting auto-initinitalization to: 0X01
[ INFO] [1491934043.919762998]: Reading back auto-initialization value
[ INFO] [1491934045.113234525]: Success: Auto init. setting is: 0X01
[ INFO] [1491934045.113579992]: Saving auto init. settings to EEPROM
[ INFO] [1491934046.151061484]: Reset filter
[ INFO] [1491934047.345137712]: Enabling AHRS stream
[ INFO] [1491934047.633095134]: Enabling Filter stream
[DEBUG] [1491934047.633524195]: 0 FILTER (0 errors)    2 AHRS (0 errors)    0 GPS (0 errors) Packets
[ INFO] [1491934047.885093747]: Enabling GPS stream
[DEBUG] [1491934048.684835100]: 0 FILTER (0 errors)    15 AHRS (0 errors)    0 GPS (0 errors) Packets
[DEBUG] [1491934049.686765172]: 0 FILTER (0 errors)    25 AHRS (0 errors)    0 GPS (0 errors) Packets
[DEBUG] [1491934050.784760332]: 0 FILTER (0 errors)    36 AHRS (0 errors)    0 GPS (0 errors) Packets
[DEBUG] [1491934051.786713634]: 0 FILTER (0 errors)    46 AHRS (0 errors)    0 GPS (0 errors) Packets
[DEBUG] [1491934052.884666764]: 0 FILTER (0 errors)    57 AHRS (0 errors)    0 GPS (0 errors) Packets

bsb808, 2017-04-11:
Sounds like a good start. Please do submit a PR with your changes. I'll look into the cpu usage.

ibaranov-cp, 2017-04-11:
@bsb808 email me (ibaranov@clearpath.ai) if you want us to send you some clearpath stuff as a small thank you (I build robot case badges, etc :)

bsb808, 2017-04-12:
@jeff-o 
I addressed some of the things you bring up...

x,y, z acceleration are reported properly. Z orientation seems stable, but takes a few moments to stabilize after the device is swung around at high speed.
>> Probably could be tuned in the complementary filter settings used by the device.  I haven't exposed all those settings through the driver, but certainly could be done.  The examples in the SDK provide examples for implementing almost all the functionality.

when canceled the node escalates to SIGTERM every time.
>> I wasn't closing the serial port ;)  Should be fixed.

there are references to "gx4" in the launch file that are easily fixed.
>> Probably should rename the package to a more general name if folks find it useful for more than just the GX5-45

on launch the node still reports "enabling GPS stream" though there is no GPS topic published.
>> Fixed

it seems to be nice and stable at 100Hz, nice!
>> Even though the "base rate" of the AHRS is 1000 Hz, the most I could get using the GX5-45 was about 220 Hz.

resource usage seems high at 100% CPU, but only 0.7% memory. Causes my computer fans to spin up.
>> Did a little work to address this, but didn't test exhaustively.  The spin rate is now a function of the message update rates, so it should be a bit more efficient.  I'm interested what your testing shows at 100Hz with the updates.

all covariance values are 0.0. I'm not sure we can trust it THAT much ;)
>> Yes, I'll open a separate issue on this.  Since we use the GX5-45, we really only pay attention to the covariance of the filter output.  If you are plumbing this into a navigation stack you might use static estimates of the covariance.  Personally I've found that this type of 'by-hand' tuning is almost always necessary.

for the sake of differentiating from other potential onboard IMU devices, make the topic name a parameter and/or change the default to something more descriptive like "gx5-imu"
>> Perhaps I'm misunderstanding the request, but couldn't you just remap the published topic(s)?

I'm glad this is getting some use beyond just our own projects, so happy to make these types of tweaks.  Even more happy to receive PR's.

bsb808, 2017-04-18:
Thank you for the pull request.  I merged the requests and removed the development branch.  Enjoy.

Brian

closed 2017-04-18 by bsb808