ros-drivers/microstrain_mips #17: gx5_link reference frame location

Enmar123, 2019-05-23:
I would like to know where on an IMU the "gx5_link" reference frame is located to build an accurate transform to it. Not all Lord IMU's are shaped the same, is there a general rule for locating this reference frame on your model or is the location unique to every model?

My IMU: 3DM-CX5-25

tonybaltovski, 2019-05-24:
As far as I know, all IMUs have their axes labelled so use that as reference.

bsb808, 2019-05-24:
I believe the discussion on issue #4 may be what you are looking for - true?

We really should create some documentation on these issues to avoid confusion.  I will try to make some time, but would be very thankful if others could help with how they are putting the devices to use.

Enmar123, 2019-05-28:
I saw the reference frame discussion on issue #4 and it was quite helpful for determining the correct IMU orientation. I have also decided to use the axes marker for the translation component of my transform, is this a good idea?