ros-drivers/microstrain_mips #24: data is not steady when when imu is in rest position

rajrajsahil, 2019-07-25:
Imu is on the robot. When the robot is rotating IMU give value close to accurate but when we stop the robot for some time IMU output(yaw) keep increasing.

samkys, 2019-09-08:
Has anyone understood what is causing this? I am seeing a very similar problem. When using the MIPs software provided by Lord Sensing, the sensor is very responsive. When using this driver the orientation is not very responsive or it responds very delayed as above. If I rotate the sensor around 180* about Z and stop the orientation keeps moving.

samkys, 2019-09-09:
@rajrajsahil It has to do with Gyro Bias I am pretty sure. It also appears that Gyro bias calculation only appears to affect the KF filtered data?

Pull from my master branch at: https://github.com/samkys/microstrain_mips/commits/master
Then build it and run the node with 

    `<param name="publish_filtered_imu" value="true" type="bool" />`

Then when you run the node, run a `rosservice call /gx5/gyro_bias_capture` while the IMU is stationary. After this visualize orientation in Rviz using `/gx5/filtered/imu/data`

This worked for me as of now!

indraneelpatil, 2019-09-24:
@samkys  Thanks for your detailed steps. Unluckily for me it didn't solve the problem. The Yaw drifts randomly in positive or negative directions while moving along a straight line.  I repeated the same experiment with another manufacturer IMU and I did not observe this drift.

 Any new updates on this thread?
[image]

indraneelpatil, 2019-09-24:
@samkys Update! Looks like your fix is working, please ignore my previous post. The IMU Yaw is much more stable now!

samkys, 2019-11-22:
Glad to hear! Good luck, and if anything changes let me know.