ros-drivers/microstrain_mips #26: [BUG] Infinite while loop on -15 or -25

samkys, 2019-09-09:
The error occurs if you set the `device_setup` param to be false so you listen for existing streams and you try calling the `dynamics_mode` service.

```
  // Get vehicle dynamics mode
  bool Microstrain::get_dynamics_mode(std_srvs::Trigger::Request &req,
      std_srvs::Trigger::Response &res)
  {
    if (GX5_15 == true || GX5_25 == true)
    {
      ROS_INFO("Device does not support this feature");
      res.success = false;
      return true;
    }

    readback_dynamics_mode = 0;
    while (mip_filter_vehicle_dynamics_mode(&device_interface_,
        MIP_FUNCTION_SELECTOR_READ, &readback_dynamics_mode) != MIP_INTERFACE_OK) {}

    ROS_INFO("Vehicle dynamics mode is: %d\n", dynamics_mode);

    res.success = true;
    return true;
  }
```

If `device_setup` is false, `GX5_15` and `GX5_25` are both false and the while loop does not have a timeout.

samkys, 2019-09-09:
I can fix this quick and submit a PR .

tonybaltovski, 2019-10-07:
Merged, thanks @samkys

closed 2019-10-07 by tonybaltovski