ros-drivers/microstrain_mips #28: Proposal to Re-License

bsb808, 2019-10-07:
TL;DR Version
-------------------

I propose that we re-license this project, changing from [GPL v3](https://www.gnu.org/licenses/gpl-3.0.en.html) to [MIT](https://choosealicense.com/licenses/mit/).

If you are a contributor - we need to record your explicit permission to change the license - please reply to this post.

-------------------

This proposal comes about after conversations with LORD Microstrain, the commercial company that produces the Microstrain line of inertial sensors.  The company is interested in supporting this driver (making contributions and updating the libraries to more modern versions of their interface, etc.)   The GPL v3 license is an impediment to their contributions because of the copyleft aspect of the license; it is perceived that their commercial customers would need to release the source code of proprietary products.  Moving to the MIT, a permissive license, would allow they to contribute to an open-source project that may have more value users of the devices.

Personally,  I believe the advantages of having the maker of the devices contribute to the ROS driver outweigh any consequences of moving to a permissive license.  Naively, I didn't give any thought to this when I started the project.  Also, to be clear, I don't work for LORD Microstrain, I won't benefit financially from this change and I don't have any plans to use this code in a proprietary project; I just want the project to benefit from some paid software developers intimately familiar the devices.

The way this would work is as follows:

1. All [contributors](https://github.com/ros-drivers/microstrain_mips/graphs/contributors) to the project would provide their explicit permission to change the license.  Contributors should provide this permission by responding to this issue.   We will check that record into the repo as a text file if/when we re-license.
    * If a contributor fails to respond or explicitly denies permission, we could consider removing their contribution(s) contributions from any re-licensed versions, but I'm really hoping that isn't an issue we have to deal with.
2. If we achieve consensus, we would make a final release of the GPL v3 licensed code.
3. Then we would change the license in the source to MIT and create another release with the new license.

If the community agrees, I'd like to make this change by the end of October.  

Comments are welcome as this is not a process I've ever been through.  The goal is to have the best open-source support for this class of devices.

wxmerkt, 2019-10-08:
Excited to see this - and very happy to agree. My contributions have been very minor, and I am happy to give explicit permission for the relicensing.

tonybaltovski, 2019-10-08:
I also grant permission for my minor changes.

bsb808, 2019-10-08:
Thank you @wxmerkt and @tonybaltovski  for the quick replies.  Hopefully we can get in touch with all the contributors.

tonybaltovski, 2019-10-08:
@ssubramaniam-cpr (@shreyasubbu) and @ljazzal where co-op students at Clearpath.  Similarly, @jeff-o works at Clearpath.

shreyasubbu, 2019-10-08:
Sounds good! I give my permission for the re-licensing.

ljazzal, 2019-10-08:
I also grant permission for the relicensing.

mglord, 2019-10-09:
Hi all!

I am on the software team at LORD Microstrain and I just wanted to reach out and thank you all for contributing to this driver! We appreciate everyone stepping in to help one another in the absence of an officially supported solution!

We are invested in expanding the driver to support more current products and functionality, as well as keeping it up-to-date as new products and features are developed and released. Using this as a base will allow us to provide this while maintaining backwards compatibility with your current implementations.

We are planning to utilize [MSCL](https://github.com/LORD-MicroStrain/MSCL), our open-source API built for simplified communication with our sensors, to take advantage of functionality that has already been implemented and allow us to keep behavior, features, and bug fixes consistent across all supported Microstrain software.

As Brian outlined above, the GPL v3 license is problematic for many of our customers who may want to utilize this driver in their products without making the source available publicly - in this regard the MIT license is much more flexible. Although we want to provide customers the option of keeping their code private, we remain committed to keeping all our work on this driver open-source and free - just as with MSCL, we feel this is the best way to provide the most flexible, usable solution.

Feel free to reach out to me with any questions, concerns, suggestions, features you'd like to see supported, etc. - we'd love to hear from you!

Regardless of whether or not you agree to re-license your contributions, thank you for helping us to support the Microstrain community!

tonybaltovski, 2019-10-09:
@mglord I had a student take a look at MSCL API but there was issues with it using boost what wasn't supported by many of the targeted Ubuntu LTS versions that ROS uses.

jeff-o, 2019-10-09:
Fine with me!

mglord, 2019-10-10:
@tonybaltovski Thanks for bringing that to my attention - it's definitely something we'll look into/try to resolve before moving forward with MSCL!

mglord, 2019-11-22:
Hello again! Thank you all for responding so quickly and granting permission to re-license your contributions!

I think there are only a few contributors we haven't yet heard from - @pvechersky and @samkys, it would be great to get your input either way (granting or denying)! If you're not willing to re-license your contributions under MIT or have any questions or concerns regarding the switch feel free to either post here or email me directly at [melissa_gill@lord.com](melissa_gill@lord.com).

pvechersky, 2019-11-22:
Sorry for the late response, this slipped through the cracks somehow.

This sounds great! I also grant permission for the relicensing.

samkys, 2019-11-22:
I grant permission to move any and all of my contribution to the MIT license.

Also sorry for the late response. I am totally in favor of this move! If there is a way to keep me informed as to the future changes that would be great.

mglord, 2019-12-03:
Fantastic! Thanks so much for your permission to re-license!

Moving forward our goals are to:
- Release a version that replaces the MIP SDK with MSCL with no changes to the public interface for easy transition (we've found a solution to the boost compatibility issues)
- Expand documentation, quick start guides
- Ensure support for all MIP devices with existing functionality
- Add support for any outstanding data fields, commands
- Maintain release notes for all new features and bug fixes in stable versions

Once the license is updated we will fork the repository under the LORD-MicroStrain account - I'll post a link here when that happens. If there is an additional way we can help keep you informed of changes we're open to suggestions!

Thank you all for your contributions and help in supporting the community!

bsb808, 2019-12-18:
Thank you all for taking the time to respond to this issue and granting your permission to re-license.  I think having industry support for the driver will benefit everyone in the future.

One other questions for @mglord - Once you make your fork, consider if you should also be involved in maintaining the documentation at http://wiki.ros.org/microstrain_3dm_gx5_45 and https://index.ros.org/p/microstrain_mips/github-ros-drivers-microstrain_mips/#melodic

I will take the steps outlined at the top of this thread.

bsb808, 2019-12-18:
Closing the issue as resolved.

closed 2019-12-18 by bsb808

mglord, 2019-12-18:
@bsb808 It seems like we should be involved in maintaining any documentation you're currently maintaining - thanks for bringing these to my attention! I'll reach out to you to coordinate access and ownership!