ros-drivers/microstrain_mips #31: GX5-25 IMU: Invalid timestamp, not using GPS Correlation Timestamp

jeremyroy, 2020-01-08:
**Problem**

The timestamp in the IMU ROS message being published from this node is invalid and non-synchronous.  After some investigation, we realized that the issue is originating from how the timestamp is being generated.  When a new ARHS packet is received from the GX5 IMU (in our case the GX5-25), the node isn't parsing the "GPS Correlation Timestamp" message, which includes a timestamp generated internally by the GX5.  Instead, the node is using ROS::Time::now(), whose accuracy is subject to when the operating system woke up the node after a new MIPS packet becomes available.

**Action items**

1. Enable the "GPS Correlation Timestamp (0x80, 0x12)" MIPS message to publish at the same rate as the IMU data
2. Implement parsing of the "GPS Correlation Timestamp (0x80, 0x12)" MIPS message within the node (if not already implemented)
3. Use the GPS Correlation Timestamp in the ROS IMU message header.
4. For the GX5-25 which doesn't have a built-in GNSS receiver, allow the option for sending a "GPS Time Update (0x01, 0x72)" command to set and re-calibrate the GX5-25's internal GPS time.  As per the GX5-25 Data Communications Protocol Manual, there should be an option of sending the external GPS time periodically: "It is recommended that this update command be sent once per second"

jeremyroy, 2020-01-20:
This issue might be related to issue #32.  Our observed non-synchronicity could also be caused by the parser dropping ARHS packets.

CaptKrasno, 2021-02-25:
You may want to switch to this fork:  https://github.com/LORD-MicroStrain/ROS-MSCL

It actively maintained by the developer and the time synchronization seems to work well.  I created a pull request (that was just merged) that publishes the GPS correlation timestamp as a ROS message so you can check the PPS status and time synchronization.  

-Kris