ros-drivers/microstrain_mips #32: GX5-25 Parser error

hughcorley, 2020-01-08:
When running the launch file in this repo, the 3DM-GX5-25 appears to give a few warnings as well as a parser error. I've attached screenshots of rqt_robot_monitor while running said launch file. Any ideas why this might be happening?

[image]
[image]
[image]

weicwc, 2020-01-12:
I am encountering the same issue as well. This caused the published imu/data rate to be below the expected rate. Attempting to run at different baudrate still result in parser error.

jeremyroy, 2020-01-20:
@weicwc Your IMU rate problem might also be related to issue #31.